cambel/ur3 #3: ubuntu18.04 ROS_melodic  catkin_make problem

Alan963852741, 2020-08-13:
Hi, my friend. I try to use your code on Ubuntu18.04 which the ros version is melodic, but your version is kinect in Ubuntu16.04. When I run "catkin_make", I got these errors.It looks like C++ problem. So is there anyone can solve it?
Thanks!

/home/alan/ros_ws/src/robotiq/robotiq_gazebo/src/mimic_joint_plugin.cpp: In destructor ‘virtual gazebo::MimicJointPlugin::~MimicJointPlugin()’:
/home/alan/ros_ws/src/robotiq/robotiq_gazebo/src/mimic_joint_plugin.cpp:50:3: error: ‘DisconnectWorldUpdateBegin’ is not a member of ‘gazebo::event::Events’
   event::Events::DisconnectWorldUpdateBegin(this->updateConnection);
   ^
/home/alan/ros_ws/src/robotiq/robotiq_gazebo/src/mimic_joint_plugin.cpp: In member function ‘void gazebo::MimicJointPlugin::UpdateChild()’:
/home/mhy/ros_ws/src/robotiq/robotiq_gazebo/src/mimic_joint_plugin.cpp:170:39: error: ‘class gazebo::physics::World’ has no member named ‘GetPhysicsEngine’
   static ros::Duration period(world_->GetPhysicsEngine()->GetMaxStepSize());

cambel, 2020-08-14:
Hi @Alan963852741 

Yeah, this package initially just supported ROS Kinetic (Ubuntu 16.04) but I have migrated to ROS Melodic (Ubuntu 18.04). The problem you have is not directly related to this repository but rather it is because of 2 of its dependencies that need to upgrade their Gazebo-related libraries.

Anyway, in order to make it work, you can just switch the following git repositories, **gazebo_ros_link_attacher** and **robotiq**, to the "melodic-devel" branch, then compile.

```
cd ~/ros_ws/src/gazebo_ros_link_attacher \
     && git checkout melodic-devel \
     && cd ~/ros_ws/src/robotiq \
     && git fetch origin && git checkout melodic-devel
```

Alan963852741, 2020-08-14:
@cambel Thanks for your help!.It worked.

closed 2020-08-14 by Alan963852741