cambel/ur3 #4: Have nothing after rosrun ur3_gazebo send_joints.py

Diankuang-Wu, 2020-09-01:
Hi, when input the command **roslaunch ur3_gazebo ur3_cubes.launch** and **rosrun ur3_gazebo send_gripper.py --value 0.5**, it worked, no problem, but input **rosrun ur3_gazebo send_joints.py**, have nothing, no any message output in the terminal, only the cursor flashing
[image]
A similar issue below:
source from https://github.com/ros-industrial/universal_robot, input command:
roslaunch ur_gazebo ur5.launch limited:=true
roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true limited:=true
roslaunch ur5_moveit_config moveit_rviz.launch config:=true
and then 
python test_move.py (from https://github.com/ros-industrial/ur_modern_driver)
output in the terminal:
[image]
someone said that because test_move.py is executed for real UR, but I think for gazebo .py file is ok, but import *** is the key.
[image]
So, what should I do for it? Thank you~

cambel, 2020-09-01:
Hi @Diankuang-Wu 

For the first issue, you need to modify the script `send_joints.py` and specify the joint angles that you want. However, I actually do not use that script anymore. I created another script that I find more useful:
`rosrun ur_control joint_position_keyboard`
There you can control each joint independently using the keyboard, press SPACE to see how to control the robot.

For both issues, I use a prefix for the  _follow_joint_trajectory_ node, so you would need to rename the node as:

- `arm_controller/follow_joint_trajectory`  on line 126 for the test_mode.py script

- or `arm_controller/command` on line 26 for the send_joint.py script (I will update that but still I do not recommend using it)

Diankuang-Wu, 2020-09-02:
Thank you very much! And I apologize for the late reply.
Your advice is helpful, it worked.
If any other relative issue happen, maybe I will bother you again, please don't mind. Thank you again.

cambel, 2020-09-02:
Glad to hear that.
Sure, let me know if you have any issue.

closed 2020-09-02 by cambel