cambel/ur3 #6: ROSRUN question

Twilight89, 2021-01-26:
Hi, when I do the same to git this repo(not the docker one),  I can run the visualization in rviz, but got error in robot grasp(rosrun ur_control sim_controller_examples.py -m ) . 
It occurs as :
Traceback (most recent call last):
  File "/home/twilight/catkin_ws_learn/src/ur3/ur_control/scripts/sim_controller_examples.py", line 4, in <module>
    from ur_control.impedance_control import AdmittanceModel
ImportError: No module named impedance_control.

It happens in rosrun ur_control joint_position_keyboard.py:
Traceback (most recent call last):
  File "/home/twilight/catkin_ws_learn/src/ur3/ur_control/scripts/joint_position_keyboard.py", line 9, in <module>
    from ur_control.arm import Arm
  File "/home/twilight/catkin_ws_learn/src/ur3/ur_control/src/ur_control/arm.py", line 19, in <module>
    from ur_ikfast import ur_kinematics as ur_ikfast
ImportError: No module named ur_ikfast

I am a fresher in ros, hoping for your reply.
Thanks~

Twilight89, 2021-01-26:
By the way, I use melodic in 18.04, and python 2.7

cambel, 2021-01-26:
@Twilight89 Hi thank you for checking my repo.

I fixed the import problems, please update your repo with the latest version.

For the problem with ` rosrun ur_control joint_position_keyboard.py` please see the Readme again, the problem is a missing library. Please follow the instructions to install the ur_ikfast library.

Twilight89, 2021-01-26:
> @Twilight89 Hi thank you for checking my repo.
>
> I fixed the import problems, please update your repo with the latest version.
>
> For the problem with ` rosrun ur_control joint_position_keyboard.py` [...]

Appreciate for your reply. I repo the new one, and down glone ur_ikfast to /src/ur3, but it occurs another question:(both run sim_controller_examples.py and joint_position_keyboard.py)
Traceback (most recent call last):
  File "/home/twilight/catkin_ws_learn/src/ur3/ur_control/scripts/sim_controller_examples.py", line 4, in <module>
    from ur_control.arm import Arm
  File "/home/twilight/catkin_ws_learn/src/ur3/ur_control/src/ur_control/arm.py", line 23, in <module>
    from trac_ik_python.trac_ik import IK
ImportError: No module named trac_ik_python.trac_ik
I think trak_ik_python maybe a dependency but I fail to build. But i do as the readme, don't know where the problem.
hope for your reply Thanks~

Twilight89, 2021-01-26:
By the way, I cant find .py file trac_ik_python in my path, so there is no IK class.

Twilight89, 2021-01-26:
When I annotate this line  "from trac_ik_python.trac_ik import IK", the problem was solved. I can run the example. Fantastic, I love open source. Thank you very much bro.

cambel, 2021-01-27:
@Twilight89  Hi,

Thanks, IK_TRACK is another dependency, I updated the Readme with the instructions to install that dependency and some additional if you want to control the real robot. Thanks, I was able to clarify the instructions for installing this repo.

Though in the future, I would recommend Docker, it is a lot easier to use, and it would provide a working environment.

Twilight89, 2021-01-27:
> @Twilight89 Hi,
>
> Thanks, IK_TRACK is another dependency, I updated the Readme with the instructions to install that dependency and some additional if you want to control the real robot. Thanks, I [...]

It did work. Thanks a lot!

closed 2021-01-27 by cambel