cambel/ur3 #8: roslaunch ur3_gazebo ur3_cubes.launch grasp_plugin:=1

huaji98, 2021-01-28:
error loading <rosparam> tag: 
	file does not exist [/home/user/ros_ws/src/robotiq/robotiq_control/config/gripper_85_controller.yaml]
XML is <rosparam command="load" file="$(find robotiq_control)/config/gripper_85_controller.yaml"/>
The traceback for the exception was written to the log file

cambel, 2021-01-29:
Hi @huaji98 ,

Are you using the Docker environment or did you install the repository manually?

If it's the latter, make sure that the docker package `robotiq` is in your ros_ws/src directory. It should have been installed in this step `rosinstall ~/ros_ws/src /opt/ros/kinetic src/ur3/dependencies.rosinstall`

huaji98, 2021-01-31:
Thank you for your reply, I checked the robotiq package, but no gripper_85_controller.yaml there

cambel, 2021-01-31:
@huaji98 Hi,

Yeah, you are right, sorry about that, I haven't used the kinetic branch (master) in a while, so I forgot to update it.

If you update the robotiq repository it should work.

Let me know if you have any other problem

huaji98, 2021-01-31:
`rosrun ur_control sim_controller_examples.py -m`
Traceback (most recent call last):
  File "/home/user/ros_ws/src/ur3/ur_control/scripts/sim_controller_examples.py", line 4, in <module>
    from ur_control.arm import Arm
  File "/home/user/ros_ws/src/ur3/ur_control/src/ur_control/arm.py", line 22, in <module>
    from ur_pykdl import ur_kinematics
  File "/home/user/ros_ws/devel/lib/python2.7/dist-packages/ur_pykdl/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 29, in <module>
  File "/home/user/ros_ws/src/ur3/ur_pykdl/src/ur_pykdl/ur_pykdl.py", line 38, in <module>
    from ur_kdl.kdl_parser import kdl_tree_from_urdf_model
  **File "/home/user/ros_ws/src/ur3/ur_pykdl/src/ur_kdl/__init__.py", line 1, in <module>
    from .kdl_parser import (
  File "/home/user/ros_ws/src/ur3/ur_pykdl/src/ur_kdl/kdl_parser.py", line 146
    print("URDF non-fixed joints: %d;" % num_non_fixed_joints, end=' ')**
                                                                  ^
SyntaxError: invalid syntax

cambel, 2021-01-31:
Umm the problem is that the Kinematics library `ur_pykdl` does not support Python 2 anymore.

I strongly recommend you to use python 3. You can use the Dockerfile as an example to either set up your local machine or use Docker directly.

Sorry!

closed 2021-02-07 by cambel