cambel/ur3 #9: ur_control

Twilight89, 2021-01-29:
Hi~ I'm pretty fresh in ROS. I download this repo and it did work. My question is that whether this project use Moveit? Because I didn't find moveit group in code. If not use, how can this make robot arm move?
Hope for rely, thanks.

cambel, 2021-01-30:
Hi @Twilight89 ,

I don't actually use MoveIt to control the robot, I control it through ros-control. You can find examples of how I control the robot in the script `ur_control/scripts/sim_controller_examples.py`

However, now I also want to use some of the MoveIt options for planning, so I am trying to setup MoveIt to control the robot as well.

So far, I am able to control the robot using the rviz interface. Please update the repository and check the GUI interface using the command:
`roslaunch ur3e_hande_moveit_config start_moveit.launch`

cambel, 2021-02-05:
Hi @Twilight89,

If you are interested, I included the [moveit_tutorial](https://github.com/cambel/ur3/blob/master/ur_control/scripts/moveit_tutorial.py) working with my current set up.

In other to make it work, start one of the gazebo environments, such as:
`roslaunch ur3_gazebo ur3e_cubes.launch`

Then load the MoveIt configuration
`roslaunch ur3e_hande_moveit_config start_moveit.launch`

Then execute the tutorial
`rosrun ur_control moveit_tutorial.py --tutorial`

 Let me know if you have further questions

closed 2021-02-05 by cambel