cambel/ur3 #11: Can't run scripts Error: package 'ur_control' not found

Gabriel-la-flamboyance, 2021-07-19:
Hello sir, I am trying to run your repo, it works for the RVIZ ("**roslaunch ur3_description display_with_gripper.launch**") and Gazebo ("**roslaunch ur3_gazebo ur3_cubes.launch grasp_plugin:=1**"), but when I try to run the scripts it doesn't work ("**rosrun ur_control sim_controller_examples.py -m**"), it says it can't found the ur_control package, even if it is present, could you help me please? 

**[image]**
[image]

cambel, 2021-07-19:
Hi, could you provide more information of what you are trying to do?

Did you install this repository using the Docker method or from source?

Please copy paste the stack trace of the error in the terminal instead of a cropped image.

In any case, it seems a problem with the compilation or sourcing of the ROS workspace. Try to compile again the ROS workspace with `catkin build` and then source it `source your_ros_ws/devel/setup.bash`

Gabriel-la-flamboyance, 2021-07-19:
Thank you for your quick reply, I am just trying to do the same thing like you did and try to understand it so i could reproduce it myself later, 
I have Installed the repository from source, 
When i input : 
**rosrun ur_control sim_controller_examples.py -m**
the error is : 
**[rospack] Error: package 'ur_control' not found**

cambel, 2021-07-19:
Then can you try compiling with `catkin build` again and paste here the result of the compilation

Gabriel-la-flamboyance, 2021-07-19:
Now I just have to deal with the scripts issues, 
When i make "**rosrun ur_control sim_controller_examples.py -m**" for example it shows:
[image]
Which means i suppose that I have some problems with the scripts.

Gabriel-la-flamboyance, 2021-07-19:
**Traceback (most recent call last):
  File "/home/fabmanager/ros_ws/src/ur_control/scripts/sim_controller_examples.py", line 4, in <module>
    from ur_control.arm import Arm
  File "/home/fabmanager/ros_ws/src/ur_control/src/ur_control/arm.py", line 22, in <module>
    from ur_pykdl import ur_kinematics
  File "/home/fabmanager/ros_ws/devel/lib/python2.7/dist-packages/ur_pykdl/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 29, in <module>
  File "/home/fabmanager/ros_ws/src/ur_pykdl/src/ur_pykdl/ur_pykdl.py", line 38, in <module>
    from ur_kdl.kdl_parser import kdl_tree_from_urdf_model
  File "/home/fabmanager/ros_ws/src/ur_pykdl/src/ur_kdl/__init__.py", line 1, in <module>
    from .kdl_parser import (
  File "/home/fabmanager/ros_ws/src/ur_pykdl/src/ur_kdl/kdl_parser.py", line 146
    print("URDF non-fixed joints: %d;" % num_non_fixed_joints, end=' ')
                                                                  ^
SyntaxError: invalid syntax**

cambel, 2021-07-19:
Yes, it is not supposed to work with Python 2.7 anymore.

You could try to fix the syntax errors which are not many I suppose, and try again. But I suggest you try compiling with Python 3. However, that in a local environment can be very tedious to do, thus my choice of using Docker to have an isolated environment.

Gabriel-la-flamboyance, 2021-07-19:
Unless there is other problems with my VM, but my python is 3.6. I am going to use Docker as i found that it is a little bit tedious to resolve all errors.

Gabriel-la-flamboyance, 2021-07-19:
As i am not familiar with docker so it will take me some time to figure it out but for the moment i have this issue. 
"
[image]

cambel, 2021-07-29:
I have not seen that error before. Also, did you say VM? do you have your environment in a VM?
Docker is basically a VM so I would not recommend having it inside another VM.

I simplify the installation process for using this repo on Ubuntu 20.04 with ROS Noetic and Python3, you can check it out in the noetic-devel branch