cambel/ur3 #12: Replace by ur10

Papul-Ghosh, 2021-08-03:
Is it possible to change the ur3 by ur10? If so, how?

cambel, 2021-08-03:
Sure it can. I have been using it on a UR3e and UR5e without problem. The main changes only real changes to the core controller is the Forward Kinematics and Inverse Kinematics libraries. I updated the KDL library, so it can load the robot's URDF from the ros parameter server `robot_description`.

For the Inverse Kinematics, I have included the UR10 in my repo of [IKFast databases](https://github.com/cambel/ur_ikfast). Alternatively, you can use Trac-IK which loads the URDF from the parameter server, similarly to KDL.
All of those definitions are here https://github.com/cambel/ur3/blob/e2bf3dc74fddb9eb196126a7d937001d3e91fd1c/ur_control/src/ur_control/arm.py#L108-L122

I will update the code soon to have an example that does not relies on a fixed URDF. Instead it loads whatever URDF is on the `robot_description`.

For gazebo, you would need to create a launch file loading your robot with whatever additional configuration you have (gripper, cameras, environment, etc)

For MoveIt, you would could in principle use the [default official package](https://github.com/ros-industrial/universal_robot/tree/kinetic-devel/ur10_moveit_config) or create your own one following the MoveIt tutorials.

I think that covers most if not all the difference between UR robots in this repo.

cambel, 2021-08-09:
@Papul-Ghosh Hi,

I did some updates to the repository. Now it is easier to use any UR robot.

You can start gazebo for the cubes task world with the following command

> `roslaunch ur3_gazebo ur_cubes.launch ur_robot:=ur10`

Then you can connect to the robot defining the specific robot like this:
``` Python
    arm = Arm(robot_urdf='ur10')
```

And that should be all for the basics

closed 2021-10-15 by cambel