cambel/ur3 #13: ROS noetic error

osumilab, 2021-09-30:
Hello
When I did the following, I got an error
`rosdep install --from-paths src --ignore-src -r -y`

`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ur3_gazebo: Cannot locate rosdep definition for [robotiq_gazebo]
ur3_description: Cannot locate rosdep definition for [robotiq_description]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
`

When I put the robotiq-packages and openrave_catkin-package according to the error, I got the following error in catkin_make
`CMake Error at /home/osumi/catkin_ws/devel/share/openrave_catkin/cmake/openrave_catkin-extras.cmake:1 (find_package):
  By not providing "FindOpenRAVE.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenRAVE",
  but CMake did not find one.
`
`  Could not find a package configuration file provided by "OpenRAVE" with any
  of the following names:

    OpenRAVEConfig.cmake
    openrave-config.cmake

  Add the installation prefix of "OpenRAVE" to CMAKE_PREFIX_PATH or set
  "OpenRAVE_DIR" to a directory containing one of the above files.  If
  "OpenRAVE" provides a separate development package or SDK, be sure it has
  been installed.
`

In addition, is it  possible to use this package with ROS noetic?

I'm Japanese, so please forgive me for my poor words

osumilab, 2021-09-30:
[image]
[image]

cambel, 2021-09-30:
Hi, are you trying to install from source or using Docker?

osumilab, 2021-10-03:
迅速な対応ありがとうございます。日本語で失礼します。
ur3,robotiq,ur_ikfastをダウンロードし、プログラムを実行したのですが写真のようにあるはずのur_ikfastモジュールがないと出てしまいます。

また、私は実機でUR3eとHand-eを扱っています。UR3eに関しては、UR公式のDriverなどを駆使して動作させることができるのですが、Hand-eが対応しておらず動作できていません。
Rviz上で表示できるようにし、Moveitでプログラムを用いてhand-eの開閉を行えるようにしたいのですが、もしよろしければ助言をいただきたいです。ROS初心者であり組み込む方法？などが分かりません
よろしくお願い致しま
[image]
す。

osumilab, 2021-10-03:
I will also comment in English
Thank you for your prompt response. Excuse me in Japanese.
I downloaded ur3, robotiq, ur_ikfast and ran the program, but it comes out without the ur_ikfast module that should be as shown in the picture.

Also, I handle UR3e and Hand-e on the actual machine. Regarding UR3e, it can be operated by making full use of UR official driver etc., but Hand-e does not support it and it cannot be operated.
I would like to be able to display it on Rviz and programmatically open and close hand-e with Moveit, but I would like advice if you like. I'm new to ROS. So I don't know how to incorporate it.
I look forward to working with you.

cambel, 2021-10-03:
`ur_ikfast` is a Python library, make sure you follow the [instructions](https://github.com/cambel/ur_ikfast) on how to install it. Make sure that the python binary that ROS uses is the same used for installation (either Python 2 or 3).

For the UR3e + Hand-e: there is no way, that I am aware of, to operate the gripper using MoveIt directly yet. However, this repository includes a dependency that allows programmatically calls to operate the gripper independently of the motion planner. See an example of how to use it: https://github.com/cambel/ur3/blob/bfc171daa273bb1222e8883c430c434145f02a8a/ur_control/scripts/real_controller_examples.py#L52-L71

osumilab, 2021-10-03:
I successfully installed ur_ikfast and ran the script described in the README, but I got a NameError. I suspect that the error is related to the gripper. Is there any way to improve it?

Also, I would like to feed back the force and torque of the gripper in the future to control it, but is it difficult with Hand-e?
[image]

[image]

cambel, 2021-10-03:
It seems that you are missing this library https://github.com/pal-robotics/gazebo_ros_link_attacher , which is necessary for the gripper in simulation, to simulate `grasp`. I pushed a small fix to be able to ignore this error in case of not having the dependency, the rest of the robot should work except for the gripper's grasp.

> Also, I would like to feed back the force and torque of the gripper in the future to control it, but is it difficult with Hand-e?
The Force/Torque is not in the gripper, is in the UR3e, this repository supports both simulation and the real robot Force/Torque sensor. The class `arm` has a method `get_ee_wrench` which returns the low-pass filtered F/T data.

osumilab, 2021-10-03:
After installing some of the packages published here, the original driver no longer boots. Do you have any idea of such an erro
[image]
r?

cambel, 2021-10-03:
https://github.com/fmauch/universal_robot/issues/18 you need to update the universal_robot repository to its latest commit.

osumilab, 2021-10-04:
I saw # 18, but I don't know which part to fix after all. Being a ROS beginner, I don't even know which is the latest repository and I don't know how to commit.

cambel, 2021-10-04:
According to your screenshot (please copy-paste the text in the console next time), your universal_robot repository is in `/home/osumi/catkin_ws/src/universal_robot`

so, to get the latest commit, go there and update the repository using `git`, and then compile again
```shell
cd /home/osumi/catkin_ws/src/universal_robot
git pull
catkin build
```
or something similar. If you have more troubles with the driver, first ask there https://github.com/fmauch/universal_robot/issues

osumilab, 2021-10-04:
I haven't solved it so I'll ask you a question. Thank you for everything.

cambel, 2021-10-04:
Try to do a clean rebuild
```shell
cd /home/osumi/catkin_ws/
catkin clean
catkin build
```

osumilab, 2021-10-04:
I can't run it when I enter the workspace. I feel that something bad is happening

`$ cd /home/osumi/catkin_ws/`
`~/catkin_ws$ catkin clean`
`[clean] Error: The current or desired workspace could not be determined. Please run catkin clean from within a catkin workspace or specify the workspace explicitly with the --workspace option.
`

cambel, 2021-10-04:
try with `catkin_make clean`

osumilab, 2021-10-04:
I'm planning to ask  https://github.com/fmauch/universal_robot/issues a question, but when I run catkin build and start the driver, I get the following error message.

`[ERROR] [1633340254.086588640]: This controller requires a hardware interface of type 'scaled_controllers::ScaledPositionJointInterface'. Make sure this is registered in the hardware_interface::RobotHW class.`
`[ERROR] [1633340254.086612383]: Initializing controller 'scaled_pos_joint_traj_controller' failed
`

osumilab, 2021-10-05:
I don't get an answer, do you have any idea about this error?

cambel, 2021-10-05:
Can you paste the whole stack trace of executing only `roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=133.91.72.106 limited:=true kinematics_config:="${HOME}/my_robot_calibration.yaml" `

osumilab, 2021-10-05:
'''
.. logging to /home/osumi/.ros/log/448573b2-25c5-11ec-9869-8bf2e7b1eb95/roslaunch-osumi-System-Product-Name-46902.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://133.91.72.99:46479/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 500
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 500
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: noetic
 * /rosversion: 1.15.13
 * /scaled_pos_joint_traj_controller/action_monitor_rate: 20
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /scaled_pos_joint_traj_controller/state_publish_rate: 500
 * /scaled_pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /scaled_pos_joint_traj_controller/type: position_controll...
 * /scaled_vel_joint_traj_controller/action_monitor_rate: 20
 * /scaled_vel_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_vel_joint_traj_controller/constraints/elbow_joint/trajectory: 0.1
 * /scaled_vel_joint_traj_controller/constraints/goal_time: 0.6
 * /scaled_vel_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
[... 175 lines cut ...]
[ INFO] [1633433873.169866769]: Initializing urdriver
[ INFO] [1633433873.173369191]: waitForService: Service [/ur_hardware_interface/dashboard/play] is now available.
ERROR /tmp/binarydeb/ros-noetic-ur-client-library-0.3.2/src/comm/tcp_socket.cpp 99: Connection setup failed for 133.91.72.106:30002 
ERROR /tmp/binarydeb/ros-noetic-ur-client-library-0.3.2/src/comm/tcp_socket.cpp 99: Connection setup failed for 133.91.72.106:30004 
[FATAL] [1633433884.307070472]: Failed to connect to robot. Please check if the robot is booted and connected.
[ERROR] [1633433884.307193884]: Could not correctly initialize robot. Exiting
================================================================================REQUIRED process [ur_hardware_interface-2] has died!
process has died [pid 46919, exit code 1, cmd /home/osumi/catkin_ws/devel/lib/ur_robot_driver/ur_robot_driver_node __name:=ur_hardware_interface __log:=/home/osumi/.ros/log/448573b2-25c5-11ec-9869-8bf2e7b1eb95/ur_hardware_interface-2.log].
log file: /home/osumi/.ros/log/448573b2-25c5-11ec-9869-8bf2e7b1eb95/ur_hardware_interface-2*.log
Initiating shutdown!
================================================================================
[ur_hardware_interface/ur_robot_state_helper-6] killing on exit
[controller_stopper-5] killing on exit
[ros_control_stopped_spawner-4] killing on exit
[ros_control_controller_spawner-3] killing on exit
[ur_hardware_interface-2] killing on exit
[robot_state_publisher-1] killing on exit
[WARN] [1633433885.149083]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1633433885.150157]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[controller_stopper-5] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done

'''

osumilab, 2021-10-05:
thank you

```
roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=XXX limited:=true kinematics_config:="${HOME}/my_robot_calibration.yaml"
```

```
... logging to /home/osumi/.ros/log/2aa9d1b4-25a1-11ec-bada-3b5469e2de83/roslaunch-osumi-System-Product-Name-135363.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://133.91.72.99:37675/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /hardware_control_loop/loop_hz: 500
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 500
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: noetic
 * /rosversion: 1.15.13
 * /scaled_pos_joint_traj_controller/action_monitor_rate: 20
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /scaled_pos_joint_traj_controller/state_publish_rate: 500
 * /scaled_pos_joint_traj_controller/stop_trajectory_duration: 0.5
[... 199 lines cut ...]
[ INFO] [1633413471.119954488]: waitForService: Service [/controller_manager/switch_controller] is now available.
[ INFO] [1633413471.119976168]: Service available.
[ INFO] [1633413471.119984956]: Waiting for controller list service to come up on /controller_manager/list_controllers
[ INFO] [1633413471.120410961]: Service available.
[ INFO] [1633413471.302252472]: Robot mode is now RUNNING
[ INFO] [1633413471.302964174]: Robot's safety mode is now NORMAL
[INFO] [1633413471.341713]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1633413471.342695]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1633413471.343178]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1633413471.344155]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1633413471.344396]: Loading controller: pos_joint_traj_controller
[INFO] [1633413471.345405]: Loading controller: joint_state_controller
[INFO] [1633413471.360780]: Loading controller: joint_group_vel_controller
[INFO] [1633413471.362741]: Loading controller: scaled_pos_joint_traj_controller
[INFO] [1633413471.366819]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[ERROR] [1633413471.367709134]: This controller requires a hardware interface of type 'scaled_controllers::ScaledPositionJointInterface'. Make sure this is registered in the hardware_interface::RobotHW class.
[ERROR] [1633413471.367731725]: Initializing controller 'scaled_pos_joint_traj_controller' failed
[... cut ...] r own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[INFO] [1633413472.374675]: Loading controller: force_torque_sensor_controller
[INFO] [1633413472.378691]: Controller Spawner: Loaded controllers: joint_state_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1633413472.380697]: Started controllers: joint_state_controller, speed_scaling_state_controller, force_torque_sensor_controller
```

cambel, 2021-10-05:
I haven`t try ROS Noetic with the Universal Robot Driver, so I am not sure about your problem. However, this warning seems to indicate either a problem with your ros workspace (most likely) or a bug with the Noetic version of the driver.

> [ WARN] [1633413472.372527533]: SEVERE WARNING!!! A namespace collision has occured with plugin factory for class position_controllers::ScaledJointTrajectoryController. New factory will OVERWRITE [...]
[ WARN] [1633413472.372586350]: SEVERE WARNING!!! A namespace collision has occured with plugin factory for class velocity_controllers::ScaledJointTrajectoryController. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.

I suggest you add that part or the whole stack trace to your issue over [there](https://github.com/fmauch/universal_robot/issues/21)

osumilab, 2021-10-06:
https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/480
I installed the scaled controller in my workspace and it worked fine. Your advice helped me a lot. Thank you very much.