cambel/ur3 #14: About the code about compliant control in gazebo

posshe, 2021-10-13:
Hello, cambel:
I'm Hexu, and want to do some research about arm manipulation which needs compliant control. But I don't know how to realize compliant control on the gazebo. Could you please help me to contact with Cristian Camilo Beltran-Hernandez for asking  the code or the part about compliant control in gazebo? Because I can't contact him with this email address "cristianbehe@gmail.org".  The article is "Variable Compliance Control for Robotic Peg-in-Hole Assembly: A Deep Reinforcement Learning Approach".

Thank you very much

Hexu

posshe, 2021-10-14:
Thank you for the reply！It's really wonderful job about control and simulator, and I'll close the issue. 
Thank you very much.

cambel, 2021-10-14:
@posshe see these examples on how to use the compliance controller https://github.com/cambel/ur3/blob/master/ur_control/scripts/compliance_controller_examples.py