cambel/ur3 #15: MoveIt

mathgameapi, 2021-10-15:
After i execute " rosrun ur_control moveit_tutorial.py --tutorial"

I get solution found but controller failed during execution

[ INFO] [1634269755.086322848]: Loading robot model 'ur3'...
[ WARN] [1634269755.157867638, 1643.538000000]: IK plugin for group 'arm' relies on deprecated API. Please implement initialize(RobotModel, ...).
[ INFO] [1634269755.161496214, 1643.538000000]: IK Using joint shoulder_link -6.28319 6.28319
[ INFO] [1634269755.161637223, 1643.538000000]: IK Using joint upper_arm_link -6.28319 6.28319
[ INFO] [1634269755.161731493, 1643.538000000]: IK Using joint forearm_link -6.28319 6.28319
[ INFO] [1634269755.161813386, 1643.538000000]: IK Using joint wrist_1_link -6.28319 6.28319
[ INFO] [1634269755.161895847, 1643.538000000]: IK Using joint wrist_2_link -6.28319 6.28319
[ INFO] [1634269755.161986298, 1643.538000000]: IK Using joint wrist_3_link -6.28319 6.28319
[ INFO] [1634269755.162081891, 1643.539000000]: Looking in common namespaces for param name: arm/position_only_ik
[ INFO] [1634269755.163546127, 1643.541000000]: Looking in common namespaces for param name: arm/solve_type
[ INFO] [1634269755.167668683, 1643.546000000]: Using solve type Speed
[ INFO] [1634269756.195190390, 1644.570000000]: Ready to take commands for planning group arm.

----------------------------------------------------------
Welcome to the MoveIt MoveGroup Python Interface Tutorial
----------------------------------------------------------
Press Ctrl-D to exit at any time

============ Press `Enter` to begin the tutorial by setting up the moveit_commander ...

============ Planning frame: world
============ End effector link: ur3_robotiq_hande_gripper
============ Available Planning Groups: ['arm', 'gripper']
============ Printing robot state
joint_state: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: "world"
  name: [shoulder_pan_joint, shoulder_lift_joint, elbow_joint, wrist_1_joint, wrist_2_joint,
  wrist_3_joint, hande_left_finger_joint, hande_right_finger_joint]
  position: [1.566860301570573, -1.5668435411999733, 1.2568592740372768, -1.5637091385299646, -1.5637164488965385, -1.5409710760749817e-05, 1.363882040352969e-07, 1.363882040352969e-07]
  velocity: []
  effort: []
multi_dof_joint_state: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: "world"
  joint_names: []
  transforms: []
  twist: []
  wrench: []
attached_collision_objects: []
is_diff: False

============ Press `Enter` to execute a movement using a joint state goal ...

[ INFO] [1634269768.584891176, 1656.750000000]: ABORTED: Solution found but controller failed during execution
('Current pose', position: 
  x: -0.131295242951
  y: 0.357079426889
  z: 0.257162754126
orientation: 
  x: -0.700000738268
  y: -0.698015165505
  z: -0.104421863889
  w: 0.108948930659)
============ Press `Enter` to execute a movement using a pose goal ...

mathgameapi, 2021-10-15:
I am build from source.

i go through all step again still get the error.

Thanks you for yr help

cambel, 2021-10-15:
@mathgameapi please continue the discussion on the same issue

> I found that the ros_ws/src/ur3/ur3e_hande_moveit_config/launch/planning_context.launch have hard code the univerisal_robot path to /root/ros_ws. So it pop the error message.
> Now the gazebo can have the movement.

Great, thanks for noticing the source of the problem, I updated the repo again with that change, I renamed the moveit packages so you need to build your catkin workspace again.

> Sorry. I am just a ros beginner. If i want to run the sample on the ur3.
> Would you like to tell me how to do that?

I don't understand the question, could you be more specific on what you want to do?
Do you want the simulation of the UR3 instead of the UR3e?
if so, you can start the simulation of the UR3 using the following command
`roslaunch ur3_gazebo ur_gripper_85_cubes.launch ur_robot:=ur3 grasp_plugin:=1`

Do you want to use MoveIt but with the UR3 instead of the UR3e?
if so, you can start MoveIt with the following additional parameter
`roslaunch ur_gripper_85_moveit_config start_moveit.launch ur_robot=ur3`

mathgameapi, 2021-10-16:
Thank new repo works.

I would like to test the your code on real robotic arm ur3.

I would like to know how to do that ?

Thanks.

mathgameapi, 2021-10-16:
Thank you for your support
I will test it later