cambel/ur3 #16: Catkin fails build

meowatthemoon, 2021-11-04:
Hello I'm trying to use your code.

I tried for both ROS melodic and noetic, both result in the same compilation error.
I followed all the source installation steps.

git clone https://github.com/cambel/ur3
git clone https://bitbucket.org/traclabs/trac_ik.git
git clone https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git universal_robots_ros_driver
git clone -b calibration_devel https://github.com/fmauch/universal_robot.git universal_robot
cd ~/ros_ws
rosinstall ~/ros_ws/src /opt/ros/melodic src/ur3/dependencies.rosinstall
sudo apt-get update
rosdep fix-permissions
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro=melodic -y
cd ~/ros_ws/src/gazebo_ros_link_attacher
git checkout melodic-devel
cd ~/ros_ws/src/robotiq
git fetch origin && git checkout melodic-devel

However, catkin clean and catkin build do not exist, I assume these commands were for ROS kinetic.
I instead used catkin_make. This command results in the following error:

[
[image]
](url)

Any idea what may cause this?

Thanks in advance.

cambel, 2021-11-05:
Hi @meowatthemoon,
Thanks for trying out my repo.

> catkin clean and catkin build do not exist
They are part of the catkin tools but you need to install them first `sudo apt install python-catkin-tools`. `catkin_make` works similarly so it is also fine to use that.

It is a lot better if you copy-paste (and format) the logs from the terminal instead of screenshots. The problem seems to be your version of the `universal_robots_ros_driver`, it is a pain that they keep making breaking changes to their repository, but anyways, please pull the latest changes and try to compile again.

meowatthemoon, 2021-11-07:
Thank you for the reply.

I had to download ur msgs: git clone https://github.com/ros-industrial/ur_msgs.git

that was all.

sorry for the unwarranted issue.

Thanks again.

cambel, 2021-11-07:
Sure no problem

closed 2021-11-07 by cambel