cambel/ur3 #18: No active joints or end effectors found for group 'gripper'. Make sure that kinematics.yaml is loaded in this node's namespace.

ksjz, 2021-11-19:
Hello, I am running the following code:  （18.04 + melodic）
roslaunch ur3_gazebo ur_gripper_hande_cubes.launch ur_robot:=ur3e grasp_plugin:=1
roslaunch ur_hande_moveit_config start_moveit.launch

something is wrong：
 No active joints or end effectors found for group 'gripper'. Make sure that kinematics.yaml is loaded in this node's namespace.
[ INFO] [1637309659.534774206, 329.652000000]: Constructing new MoveGroup connection for group 'gripper' in namespace ''
[ INFO] [1637309659.622648508, 329.739000000]: Ready to take commands for planning group gripper.

[image]

Have you encountered it and how to solve it, thank you！

cambel, 2021-11-19:
Hi, 
Thank you for reporting this issue.

I have pushed an update to the repository. 

Now, to control the gripper from MoveIt, I don't know how to enable that option with the interactive markers as with the planning group `arm`. In any case, you can still update the position of the gripper in the `joints` tap:
[image]

And then go back to the `Planning` tap and `Plan & Execute` the motion.

ksjz, 2021-11-20:
> Hi, Thank you for reporting this issue.
>
> I have pushed an update to the repository.
>
> Now, to control the gripper from MoveIt, I don't know how to enable that option with the interactive markers as [...]

thank you

closed 2021-11-22 by cambel