cambel/ur3 #19: Can't run scripts Error: package 'ur_control' not found

ksjz, 2022-01-04:
i have a same issue:   https://github.com/cambel/ur3/issues/11.

Here is  the result of the compilation (catkin build)

--------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/ksjz/ur10e_hande_ws/devel:/home/ksjz/ros_ws/devel:/opt/ros/melodic
Workspace:                   /home/ksjz/ros_ws
--------------------------------------------------------------------------------
Build Space:        [exists] /home/ksjz/ros_ws/build
Devel Space:        [exists] /home/ksjz/ros_ws/devel
Install Space:      [unused] /home/ksjz/ros_ws/install
Log Space:         [missing] /home/ksjz/ros_ws/logs
Source Space:       [exists] /home/ksjz/ros_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------------------
[build] Found '39' packages in 0.0 seconds.                                    
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                      [ 1.6 seconds ]       
Starting  >>> controller_stopper                                               
Starting  >>> gazebo_ros_link_attacher                                         
Starting  >>> orocos_kdl                                                       
Starting  >>> robotiq_description                                              
Starting  >>> robotiq_ft_sensor                                                
Starting  >>> robotiq_gazebo                                                   
Starting  >>> robotiq_msgs                                                     
Starting  >>> robotiq_urcap_control                                            
Finished  <<< robotiq_description                        [ 3.7 seconds ]       
_______________________________________________________________________________
Warnings   << robotiq_ft_sensor:cmake /home/ksjz/ros_ws/logs/robotiq_ft_sensor/build.cmake.000.log
CMake Warning (dev) at CMakeLists.txt:54 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "robotiq_ft_sensor_gencfg" of target "rq_test_sensor"
  does not exist.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:50 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "robotiq_ft_sensor_gencfg" of target "rq_sensor" does
  not exist.
This warning is for project developers.  Use -Wno-dev to suppress it.

cd /home/ksjz/ros_ws/build/robotiq_ft_sensor; catkin build --get-env robotiq_ft_sensor | catkin env -si  /usr/bin/cmake /home/ksjz/ros_ws/src/robotiq/robotiq_ft_sensor --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/ksjz/ros_ws/devel/.private/robotiq_ft_sensor -DCMAKE_INSTALL_PREFIX=/home/ksjz/ros_ws/install; cd -
...............................................................................
_______________________________________________________________________________
Warnings   << gazebo_ros_link_attacher:cmake /home/ksjz/ros_ws/logs/gazebo_ros_link_attacher/build.cmake.000.log
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindBoost.cmake:911 (if):
[... 270 lines cut ...]
                                                                                        ^
In file included from /home/ksjz/ros_ws/src/universal_robot/ur_kinematics/include/ur_kinematics/ur_moveit_plugin.h:100:0,
                 from /home/ksjz/ros_ws/src/universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:88:
/opt/ros/melodic/include/moveit/kinematics_base/kinematics_base.h:344:31: note: declared here
   [[deprecated]] virtual void setValues(const std::string& robot_description, const std::string& group_name,
                               ^~~~~~~~~
/home/ksjz/ros_ws/src/universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp: In member function ‘virtual bool ur_kinematics::URKinematicsPlugin::initialize(const string&, const string&, const string&, const string&, double)’:
/home/ksjz/ros_ws/src/universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:174:88: warning: ‘virtual void kinematics::KinematicsBase::setValues(const string&, const string&, const string&, const string&, double)’ is deprecated [-Wdeprecated-declarations]
   setValues(robot_description, group_name, base_frame, tip_frame, search_discretization);
                                                                                        ^
In file included from /home/ksjz/ros_ws/src/universal_robot/ur_kinematics/include/ur_kinematics/ur_moveit_plugin.h:100:0,
                 from /home/ksjz/ros_ws/src/universal_robot/ur_kinematics/src/ur_moveit_plugin.cpp:88:
/opt/ros/melodic/include/moveit/kinematics_base/kinematics_base.h:344:31: note: declared here
   [[deprecated]] virtual void setValues(const std::string& robot_description, const std::string& group_name,
                               ^~~~~~~~~
cd /home/ksjz/ros_ws/build/ur_kinematics; catkin build --get-env ur_kinematics | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Finished  <<< ur_kinematics                              [ 14.9 seconds ]      
[build] Summary: 16 of 38 packages succeeded.                                  
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  5 packages succeeded with warnings.                       
[build]   Abandoned: 21 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 48.7 seconds total.                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

ksjz, 2022-01-04:
in addition: 
 in ur3/docker folder, there is no launch_docker.sh
Is this the cause of the compilation problem?

cambel, 2022-01-11:
Hi, it seems you have a problem compiling `trac_ik_lib` shown here in your trace:
```
Errors << trac_ik_lib:make /home/ksjz/ros_ws/logs/trac_ik_lib/build.make.000.log
/usr/bin/ld: cannot find -lnlopt_cxx
collect2: error: ld returned 1 exit status
make[2]: *** [/home/ksjz/ros_ws/devel/.private/trac_ik_lib/lib/libtrac_ik.so] Error 1
make[1]: *** [CMakeFiles/trac_ik.dir/all] Error 2
make: *** [all] Error 2
```
I am not sure why though. I would suggest trying to update the ros dependencies using the following commands from the workspace directory:
```
rosdep fix-permissions
rosdep update -q
rosdep install --from-paths src --ignore-src --rosdistro=$ROS_DISTRO -y
```

closed 2022-03-30 by cambel