cambel/ur3 #20: PyKDL Error

erenerdal, 2022-01-14:
Hi,

I am trying to run your repo using Ros noetic on Ubuntu 20.04. Gazebo works fine. When I start controlling the arm with the  command "rosrun ur_control sim_controller_examples.py -m", I received the following error:

**>>  rosrun ur_control sim_controller_examples.py -m**
Import ur_ikfast not available, IKFAST would not be supported without it
Traceback (most recent call last):
  File "workspace/catkin_ws/src/ur3/ur_control/scripts/sim_controller_examples.py", line 28, in <module>
    from ur_control.arm import Arm
  File "/workspace/catkin_ws/src/ur3/ur_control/src/ur_control/arm.py", line 46, in <module>
    from ur_pykdl import ur_kinematics
  File "/workspace/catkin_ws/devel/lib/python3/dist-packages/ur_pykdl/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 29, in <module>
  File "/workspace/catkin_ws/src/ur3/ur_pykdl/src/ur_pykdl/ur_pykdl.py", line 31, in <module>
    import PyKDL
ModuleNotFoundError: No module named 'PyKDL'_

I did change the kdl parts in the package.xml files under /src/ur3/ur_pykdl/ :
  <build_depend>liborocos-kdl</build_depend>
  <build_depend>liborocos-kdl-dev</build_depend>

however, no success...Do you have any idea how to solve this problem?
thanks in advance
eren

cambel, 2022-01-17:
Please check if you have these libraries installed or try to reinstall them:
``` shell
sudo apt-get install ros-noetic-urdf-parser-plugin
sudo apt-get install ros-noetic-urdfdom-py
```

erenerdal, 2022-01-17:
I have both...still, no success...

cambel, 2022-01-18:
Which version of python do you get when executing `python --version`?

erenerdal, 2022-01-18:
Python 3.8.3

cambel, 2022-01-18:
Please see if manually installing `sudo apt install python3-pykdl` does the trick for you

erenerdal, 2022-01-18:
python3-pykdl is already the newest version (1.4.0-7ubuntu1).

cambel, 2022-01-18:
Just importing `PyKDL` fails? for example, if you execute `python -c "import PyKDL"` does it fail with the same error?

erenerdal, 2022-01-18:
yes, it returns the same error...
ModuleNotFoundError: No module named 'PyKDL'

cambel, 2022-01-18:
Then, I am not sure what the cause of the problem is, PyKDL is in the library `python3-pykdl`. It seems that you may have multiple python environments and the default version has no access to the library. You can check which libraries are "available" with this `python -c "help('modules')"`

erenerdal, 2022-01-20:
thanks, you are right, it is solved after removing anaconda ....

closed 2022-03-30 by cambel