cambel/ur3 #21: Timed out waiting for /wrench/ topic

erenerdal, 2022-01-20:
after running this script
>> **rosrun ur_control sim_controller_examples.py -m**

I am always getting the following error:

Import ur_ikfast not available, IKFAST would not be supported without it
ft_sensor: True, ee_link: None, 
 robot_urdf: ur3e
[ WARN] [1642677431.911925953]: link 'robotiq_coupler' material 'flat_black' undefined.
[ WARN] [1642677431.911953314]: link 'robotiq_coupler' material 'flat_black' undefined.
[ WARN] [1642677431.913863821]: link 'robotiq_coupler' material 'flat_black' undefined.
[ WARN] [1642677431.913881508]: link 'robotiq_coupler' material 'flat_black' undefined.
connecting to: /scaled_pos_joint_traj_controller/command
[INFO] [1642677432.433762, 3.823000]: JointTrajectoryController initialized. ns: /
[INFO] [1642677433.019245, 3.844000]: GripperCommandAction initialized. ns: /
publish filtered wrench: /wrench/filtered
**[ERROR] [1642677436.037885, 3.961000]: Timed out waiting for /wrench/ topic**
Traceback (most recent call last):
  File "/hh_ws/src/ur3/ur_control/scripts/sim_controller_examples.py", line 291, in <module>
    main()
  File "/hh_ws/src/ur3/ur_control/scripts/sim_controller_examples.py", line 261, in main
    arm = Arm(
  File "/hh_ws/src/ur3/ur_control/src/ur_control/arm.py", line 108, in __init__
    self._init_ft_sensor()
  File "/hh_ws/src/ur3/ur_control/src/ur_control/arm.py", line 170, in _init_ft_sensor
    self.set_wrench_offset(override=False)
  File "/hh_ws/src/ur3/ur_control/src/ur_control/arm.py", line 243, in set_wrench_offset
    self._update_wrench_offset()
  File "/hh_ws/src/ur3/ur_control/src/ur_control/arm.py", line 174, in _update_wrench_offset
    self.wrench_offset = self.get_filtered_ft().tolist()
  File "/hh_ws/src/ur3/ur_control/src/ur_control/arm.py", line 229, in get_filtered_ft
    ft = [
  File "/hh_ws/src/ur3/ur_control/src/ur_control/arm.py", line 230, in <listcomp>
    ft[i] if abs(ft[i]) < ft_limitter[i] else ft_limitter[i]
**TypeError: 'NoneType' object is not subscriptable**

Do you have any idea? Indeed, I can move the robot with the keybord using this script:
rosrun ur_control joint_position_keyboard.py

cambel, 2022-01-20:
Hi, glad you could solve the previous issue.

This one I know. When you the define the Arm, you can expect the force torque sensor data to be available by setting the ft_sensor flag. Please see the line 261 in sim_controller_examples.py
If you are using Gazebo it should still work with the ft_sensor on, but for now try setting it as false and trying without it. I'll check later why it does not work with the sensor.

cambel, 2022-03-30:
See solution in #22

closed 2022-03-30 by cambel