cambel/ur3 #22: 'sim_controller_examples.py -m' throws: TypeError: 'NoneType' object is not subscriptable

Unn20, 2022-03-30:
After running this script:

`rosrun ur_control sim_controller_examples.py -m`

I've got this error:_

```
ft_sensor: True, ee_link: None, 
 robot_urdf: ur3e
Unknown attribute "iyx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_base_link']/inertial/inertia
Unknown attribute "izx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_base_link']/inertial/inertia
Unknown attribute "izy" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_base_link']/inertial/inertia
Unknown attribute "iyx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_left_knuckle_link']/inertial/inertia
Unknown attribute "izx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_left_knuckle_link']/inertial/inertia
Unknown attribute "izy" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_left_knuckle_link']/inertial/inertia
Unknown attribute "iyx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_right_knuckle_link']/inertial/inertia
Unknown attribute "izx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_right_knuckle_link']/inertial/inertia
Unknown attribute "izy" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_right_knuckle_link']/inertial/inertia
Unknown attribute "iyx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_left_finger_link']/inertial/inertia
Unknown attribute "izx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_left_finger_link']/inertial/inertia
Unknown attribute "izy" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_left_finger_link']/inertial/inertia
Unknown attribute "iyx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_right_finger_link']/inertial/inertia
Unknown attribute "izx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_right_finger_link']/inertial/inertia
Unknown attribute "izy" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_right_finger_link']/inertial/inertia
Unknown attribute "iyx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_left_inner_knuckle_link']/inertial/inertia
Unknown attribute "izx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_left_inner_knuckle_link']/inertial/inertia
Unknown attribute "izy" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_left_inner_knuckle_link']/inertial/inertia
Unknown attribute "iyx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_right_inner_knuckle_link']/inertial/inertia
Unknown attribute "izx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_right_inner_knuckle_link']/inertial/inertia
Unknown attribute "izy" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_right_inner_knuckle_link']/inertial/inertia
Unknown attribute "iyx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_left_finger_tip_link']/inertial/inertia
Unknown attribute "izx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_left_finger_tip_link']/inertial/inertia
Unknown attribute "izy" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_left_finger_tip_link']/inertial/inertia
Unknown attribute "iyx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_right_finger_tip_link']/inertial/inertia
Unknown attribute "izx" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_right_finger_tip_link']/inertial/inertia
Unknown attribute "izy" in /robot[@name='ur_robot_gazebo']/link[@name='robotiq_85_right_finger_tip_link']/inertial/inertia
Unknown tag "hardwareInterface" in /robot[@name='ur_robot_gazebo']/transmission[@name='robotiq_85_left_knuckle_trans']/actuator[@name='robotiq_85_left_knuckle_motor']
connecting to: /scaled_pos_joint_traj_controller/command
[INFO] [1648644789.700306, 3600.821000]: JointTrajectoryController initialized. ns: /
publish filtered wrench: /wrench/filtered
[ERROR] [1648644792.708880, 3603.709000]: Timed out waiting for /wrench/ topic
Traceback (most recent call last):
  File "sim_controller_examples.py", line 291, in <module>
    main()
  File "sim_controller_examples.py", line 263, in main
    gripper=args.gripper,  # Enable gripper
  File "/root/ros_ws/src/ros_ur3/ur_control/src/ur_control/arm.py", line 108, in __init__
    self._init_ft_sensor()
  File "/root/ros_ws/src/ros_ur3/ur_control/src/ur_control/arm.py", line 170, in _init_ft_sensor
    self.set_wrench_offset(override=False)
  File "/root/ros_ws/src/ros_ur3/ur_control/src/ur_control/arm.py", line 243, in set_wrench_offset
    self._update_wrench_offset()
  File "/root/ros_ws/src/ros_ur3/ur_control/src/ur_control/arm.py", line 174, in _update_wrench_offset
    self.wrench_offset = self.get_filtered_ft().tolist()
  File "/root/ros_ws/src/ros_ur3/ur_control/src/ur_control/arm.py", line 231, in get_filtered_ft
    for i in range(6)
  File "/root/ros_ws/src/ros_ur3/ur_control/src/ur_control/arm.py", line 231, in <listcomp>
    for i in range(6)
TypeError: 'NoneType' object is not subscriptable
```

For me looks like similar problem to the previous issue. 
`rosrun ur_control joint_position_keyboard.py`
also works fine

cambel, 2022-03-30:
Hi @Unn20,

Sorry, this is the same error as #21 
I finally sit down to fix it. 

Please update the repository to its latest commit and try again.

Unn20, 2022-04-09:
It works now ;). Thanks

closed 2022-04-09 by cambel