cambel/ur3 #23: Bugs in python examples of ur3 motion

Unn20, 2022-04-09:
I've tested ur3 motion examples included in python script.

There are still some bugs.

Running this:
`rosrun ur_control sim_controller_examples.py --circle`
Leads to an error:
```
ft_sensor: True, ee_link: None, 
 robot_urdf: ur3e
connecting to: /scaled_pos_joint_traj_controller/command
[INFO] [1649506354.883993, 200.621000]: JointTrajectoryController initialized. ns: /
publish filtered wrench: /wrench/filtered
[INFO] [1649506354.935803, 200.667000]: FTSensor successfully initialized
Traceback (most recent call last):
  File "/root/ros_ws/src/ros_ur3/ur_control/scripts/sim_controller_examples.py", line 291, in <module>
    main()
  File "/root/ros_ws/src/ros_ur3/ur_control/scripts/sim_controller_examples.py", line 282, in main
    circular_trajectory2()
  File "/root/ros_ws/src/ros_ur3/ur_control/scripts/sim_controller_examples.py", line 188, in circular_trajectory2
    traj = traj_utils.circunference(p1, p2, steps, axes=[0,2,1])
AttributeError: module 'ur_control.traj_utils' has no attribute 'circunference'
Exception in thread /joint_states:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_pubsub.py", line 185, in robust_connect_subscriber
    conn.receive_loop(receive_cb)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 846, in receive_loop
    self.close()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 858, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'

```

Running this:
`rosrun ur_control sim_controller_examples.py --grasp_naive`
Leads to an error:
```
ft_sensor: True, ee_link: None, 
 robot_urdf: ur3e
connecting to: /scaled_pos_joint_traj_controller/command
[INFO] [1649506467.742772, 309.963000]: JointTrajectoryController initialized. ns: /
publish filtered wrench: /wrench/filtered
[INFO] [1649506467.793453, 310.009000]: FTSensor successfully initialized
Traceback (most recent call last):
  File "/root/ros_ws/src/ros_ur3/ur_control/scripts/sim_controller_examples.py", line 291, in <module>
    main()
  File "/root/ros_ws/src/ros_ur3/ur_control/scripts/sim_controller_examples.py", line 278, in main
    grasp_naive()
  File "/root/ros_ws/src/ros_ur3/ur_control/scripts/sim_controller_examples.py", line 100, in grasp_naive
    arm.gripper.open()
AttributeError: 'NoneType' object has no attribute 'open'
Exception in thread /scaled_pos_joint_traj_controller/follow_joint_trajectory/result:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_pubsub.py", line 185, in robust_connect_subscriber
    conn.receive_loop(receive_cb)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 846, in receive_loop
    self.close()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 858, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'
```

Running this:
`rosrun ur_control sim_controller_examples.py --grasp_plugin`
Leads to an error:
```
ft_sensor: True, ee_link: None, 
 robot_urdf: ur3e
connecting to: /scaled_pos_joint_traj_controller/command
[INFO] [1649506525.039055, 365.015000]: JointTrajectoryController initialized. ns: /
publish filtered wrench: /wrench/filtered
[INFO] [1649506525.090128, 365.053000]: FTSensor successfully initialized
Traceback (most recent call last):
  File "/root/ros_ws/src/ros_ur3/ur_control/scripts/sim_controller_examples.py", line 291, in <module>
    main()
  File "/root/ros_ws/src/ros_ur3/ur_control/scripts/sim_controller_examples.py", line 280, in main
    grasp_plugin()
  File "/root/ros_ws/src/ros_ur3/ur_control/scripts/sim_controller_examples.py", line 115, in grasp_plugin
    arm.gripper.open()
AttributeError: 'NoneType' object has no attribute 'open'
```

cambel, 2022-04-09:
Right, for the first case, the code is a bit outdated so it seems I changed the usage of the `traj_utils.circumference`. I will check it later.

For the other two cases, you need to also specify that you want to use the gripper with the parameter `--gripper`. You may still encounter issues, the plugins need to be set in the gazebo launch file. I will also try to check that later.

cambel, 2022-04-10:
@Unn20 
Okay, I checked the three cases and now they work as intended.
Grasping with gazebo mechanism (usually it does not grasp anything)
`roslaunch ur3_gazebo ur_gripper_85_cubes.launch ur_robot:=ur3 grasp_plugin:=0` Gazebo launcher 
`rosrun ur_control sim_controller_examples.py --grasp_naive --gripper`

Grasping with [plugin](https://github.com/pal-robotics/gazebo_ros_link_attacher)
`roslaunch ur3_gazebo ur_gripper_85_cubes.launch ur_robot:=ur3 grasp_plugin:=1` Gazebo launcher with plugin enabled
`rosrun ur_control sim_controller_examples.py --grasp_plugin --gripper`

Example of circular trajectory:
`rosrun ur_control sim_controller_examples.py --circle`
Note that if you are using the Docker environment, though it is ROS melodic, the default python version is python3.
You need to install a **tf** [hack](https://github.com/rospypi/simple) like this 
 `pip install --extra-index-url https://rospypi.github.io/simple/ tf2_ros`

closed 2022-12-27 by cambel