cambel/ur3 #24: DOCKER_RUNTIME variable is not set in docker-compose

ACTP0H0M, 2022-04-14:
Hello, I get the following error when trying to install ur3 repository with Docker:

Command:

`$ docker-compose build`

Error:

`WARNING: The DOCKER_RUNTIME variable is not set. Defaulting to a blank string.
ERROR: The Compose file './docker-compose.yml' is invalid because:
Unsupported config option for services.ros_ur3: 'runtime'`

Any idea what this argument should be or where is it read from?

cambel, 2022-04-14:
Hi @ACTP0H0M,

About that, you need to set that parameter in your local machine. Either temporarily as
`export DOCKER_RUNTIME=nvidia` if you have NVIDIA drivers or `DOCKER_RUNTIME=runc` if not.
You can set that line in  your `~/.bashrc` profile file to have it automatically set every time you open a new terminal

ACTP0H0M, 2022-04-28:
Hi, thank you for the reply, it helped to solve this issue

closed 2022-04-28 by ACTP0H0M

NigerChel, 2023-07-27:
Hello, first I want to thank for your excellent work in this repository.

I would like to reopen this issue since I have the same problem trying to use "docker-compose build", even I set up export DOCKER_RUNTIME=nvidia or export DOCKER_RUNTIME=runc in  ~/.bashrc, it gives the following issue:
[image]

I wonder if there are other options as "runc" that can solve it. Unfortunately, I don't know much about drivers.

I am using the melodic branch and Ubuntu 18.04 in a ThinkPad t420.

Thank you so much!

cambel, 2023-07-31:
Hi @NigerChel 

Do you have the latest version of docker-compose?
The problem seems unrelated to this package. Rather is a problem with docker-compose.
See here https://github.com/docker/compose/issues/6436

NigerChel, 2023-07-31:
Hi @cambel
 
Thanks for your response, after upgrading docker-compose it fixed the problem by using the following lines:

$ sudo apt-get remove docker-compose
$ sudo curl -L "https://github.com/docker/compose/releases/download/1.23.2/docker-compose-$(uname -s)-$(uname -m)" -o /usr/local/bin/docker-compose
$ sudo chmod +x /usr/local/bin/docker-compose
$ sudo ln -s /usr/local/bin/docker-compose /usr/bin/docker-compose

Now I faced with this problem of dependencies, it seems it fails the "docker-compose build"  in the Dockerfile in these lines:

# Updating ROSDEP and installing dependencies
RUN cd ~/ros_ws \
    && rosinstall ~/ros_ws/src /opt/ros/$ROS_DISTRO src/ros_ur3/dependencies.rosinstall \
    && apt-get update \
    && rosdep fix-permissions \
    && rosdep update \
    && rosdep install --from-paths src --ignore-src --rosdistro=$ROS_DISTRO -y \
    && rm -rf /var/lib/apt/lists/* 

[image]

[image]

[image]

Any idea will be very helpful !!
 
Thanks in advance.

cambel, 2023-08-01:
Melodic is no longer maintained, so I would suggest using Noetic.

If you still want to try Melodic, see this post to fix your current problem.
https://answers.ros.org/question/400079/error-the-following-packagesstacks-could-not-have-their-rosdep-keys-resolved-to-system-dependencies/