cambel/ur3 #25: robotiq_control : no definition of python-pymodbus

qdl49, 2023-05-23:
Hello, I have a problem to install dependecies :

qld@Ubuntu-20:~/catkin_ws$ rosdep install --from-paths src --ignore-src --rosdistro=noetic -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
robotiq_control: No definition of [python-pymodbus] for OS version [focal]

I installed python3-pymodbus but still not working.
Anyone can help me ? Thank you

SuperLuu7, 2023-07-03:
I encountered the same problem, did you solve it?

cambel, 2023-07-04:
@SuperLuu7 @qdl49 Can you check that the `robotiq` package is on the `noetic-devel` branch?

SuperLuu7, 2023-07-04:
> @SuperLuu7 @qdl49 Can you check that the `robotiq` package is on the `noetic-devel` branch?

Thank you for your reply. First when I check with `rospack list | grep robotiq` command, I found that there is no `robotiq` on my `ros-noetic`. Then I manually installed one. how should i solve this problem
Install command:  
`git clone https://github.com/ros-industrial/robotiq.git`
`rosdep install --from-paths src --ignore-src --rosdistro noetic -y`
 `rospack list | grep robotiq` query again, and there is `robotiq` on my `ros-noetic`. But I still get the error:
```
[ 97%] Built target robotiq_3f_rviz
/home/lilulu/ros_test5/src/robotiq/robotiq_3f_gripper_control/src/robotiq_3f_gripper_can_node.cpp: In function ‘int main(int, char**)’:
/home/lilulu/ros_test5/src/robotiq/robotiq_3f_gripper_control/src/robotiq_3f_gripper_can_node.cpp:144:41: warning: ‘virtual bool can::DriverInterface::init(const string&, bool)’ is deprecated: provide settings explicitly [-Wdeprecated-declarations]
  144 |   while(!driver_->init(can_device, false))
      |                                         ^
In file included from /opt/ros/noetic/include/socketcan_interface/asio_base.h:4,
                 from /opt/ros/noetic/include/socketcan_interface/socketcan.h:4,
                 from /home/lilulu/ros_test5/src/robotiq/robotiq_3f_gripper_control/include/robotiq_3f_gripper_control/robotiq_3f_gripper_can_client.h:29,
                 from /home/lilulu/ros_test5/src/robotiq/robotiq_3f_gripper_control/src/robotiq_3f_gripper_can_node.cpp:26:
/opt/ros/noetic/include/socketcan_interface/interface.h:180:65: note: declared here
  180 |     [[deprecated("provide settings explicitly")]]  virtual bool init(const std::string &device, bool loopback) = 0;
      |                                                                 ^~~~
[ 98%] Linking CXX executable /home/lilulu/ros_test5/devel/lib/robotiq_3f_gripper_control/robotiq_3f_gripper_can_node
[ 98%] Built target robotiq_3f_gripper_can_node
make: *** [Makefile:141: all] Error 2
**Invoking "make -j12 -l12" failed**
```
how should I solve this problem?

cambel, 2023-07-05:
@SuperLuu7 

`rospack list` lists the installed packages. The error that you had initially was related to the building of that package, so it won't show up.  Now, I use a custom version of that library which is already somewhere in your `catkin_ws` so you need to remove the package `ros-industrial/robotiq`.

What you need to check is the git branch of that `robotiq` package inside your catkin.

I think I understand the problem now. I forgot to update the installation instructions for Noetic so you had the dependencies issue due to executing instructions for Melodic.

Please try to follow the new [instructions](https://github.com/cambel/ur3/wiki/Compile-from-source) that are for Noetic from the beginning.

SuperLuu7, 2023-07-05:
> Thanks for your reply, after compiling from source according to your [instructions](https://github.com/cambel/ur3/wiki/Compile-from-source), building the whole workspace with command `catkin build `gives the following error:
```
_______________________________________________________________________________
Errors     << catkin_tools_prebuild:cmake /home/lilulu/ros_test6/logs/catkin_tools_prebuild/build.cmake.000.log
CMake Error at /opt/ros/noetic/share/catkin/cmake/empy.cmake:30 (message):
  Unable to find either executable 'empy' or Python module 'em'...  try
  installing the package 'python3-empy'
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/all.cmake:164 (include)
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

cd /home/lilulu/ros_test6/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/cmake /home/lilulu/ros_test6/build/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/lilulu/ros_test6/devel/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/home/lilulu/ros_test6/install; cd -

...............................................................................
Failed     << catkin_tools_prebuild:cmake                [ Exited with code 1 ]
Failed    <<< catkin_tools_prebuild                      [ 0.6 seconds ]       
Abandoned <<< gazebo_ros_link_attacher                   [ Unrelated job failed ]
Abandoned <<< robotiq_control                            [ Unrelated job failed ]
Abandoned <<< robotiq_description                        [ Unrelated job failed ]
Abandoned <<< robotiq_ft_sensor                          [ Unrelated job failed ]
Abandoned <<< robotiq_gazebo                             [ Unrelated job failed ]
Abandoned <<< robotiq_msgs                               [ Unrelated job failed ]
Abandoned <<< robotiq_urcap_control                      [ Unrelated job failed ]
Abandoned <<< trac_ik                                    [ Unrelated job failed ]
Abandoned <<< trac_ik_lib                                [ Unrelated job failed ]
Abandoned <<< trac_ik_examples                           [ Unrelated job failed ]
Abandoned <<< trac_ik_kinematics_plugin                  [ Unrelated job failed ]
Abandoned <<< trac_ik_python                             [ Unrelated job failed ]
Abandoned <<< universal_robots                           [ Unrelated job failed ]
Abandoned <<< ur3_description                            [ Unrelated job failed ]
Abandoned <<< ur3_gazebo                                 [ Unrelated job failed ]
Abandoned <<< ur_control                                 [ Unrelated job failed ]
Abandoned <<< ur_dashboard_msgs                          [ Unrelated job failed ]
Abandoned <<< ur_description                             [ Unrelated job failed ]
Abandoned <<< ur10_moveit_config                         [ Unrelated job failed ]
Abandoned <<< ur10e_moveit_config                        [ Unrelated job failed ]
Abandoned <<< ur16e_moveit_config                        [ Unrelated job failed ]
Abandoned <<< ur3_moveit_config                          [ Unrelated job failed ]
Abandoned <<< ur3e_moveit_config                         [ Unrelated job failed ]
Abandoned <<< ur5_moveit_config                          [ Unrelated job failed ]
Abandoned <<< ur5e_moveit_config                         [ Unrelated job failed ]
Abandoned <<< ur_gazebo                                  [ Unrelated job failed ]
Abandoned <<< ur_gripper_85_moveit_config                [ Unrelated job failed ]
Abandoned <<< ur_hande_moveit_config                     [ Unrelated job failed ]
Abandoned <<< ur_handeye_calibration                     [ Unrelated job failed ]
Abandoned <<< ur_kinematics                              [ Unrelated job failed ]
Abandoned <<< ur_pykdl                                   [ Unrelated job failed ]
Abandoned <<< ur_robot_driver                            [ Unrelated job failed ]
Abandoned <<< ur_calibration                             [ Unrelated job failed ]
[build] Summary: 0 of 34 packages succeeded.                                   
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 33 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 0.7 seconds total.  

```
So I build the whole workspace with the command `catkin_make -DPYTHON_EXECUTABLE=/usr/bin/python3`, but I still get the following error:
```
[ 95%] Built target gazebo_ros_link_attacher
In file included from /home/lilulu/ros_test6/src/universal_robots_ros_driver/ur_robot_driver/src/hardware_interface.cpp:30:
/home/lilulu/ros_test6/src/universal_robots_ros_driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h: In constructor ‘ur_driver::HardwareInterface::HardwareInterface()’:
/home/lilulu/ros_test6/src/universal_robots_ros_driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:283:20: warning: ‘ur_driver::HardwareInterface::cartesian_pose_command_’ will be initialized after [-Wreorder]
  283 |   urcl::vector6d_t cartesian_pose_command_;
      |                    ^~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
      |                                                  ^~~~
make[2]: *** [universal_robots_ros_driver/ur_robot_driver/CMakeFiles/ur_robot_driver_plugin.dir/build.make:76: universal_robots_ros_driver/ur_robot_driver/CMakeFiles/ur_robot_driver_plugin.dir/src/hardware_interface.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:10024: universal_robots_ros_driver/ur_robot_driver/CMakeFiles/ur_robot_driver_plugin.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j12 -l12" failed

```
How can I solve it?

cambel, 2023-07-07:
The first issue is that you have a missing dependency, so you should install it:
`sudo apt-get install python3-empy`

The second problem is the package `ur_robot_driver` is it in the right branch recommended by the devs?

closed 2023-11-13 by cambel