cambel/ur3 #27: Running the gripper on real equipment

PartyKusZ, 2023-07-17:
Hi,
I've been trying to get the ability to control the grapple on real equipment for a while now, but I'm not able to and I keep failing all the time. Please give me some advice. 
ROS melodic 
Ubuntu 18.04
gripper 85
ur3 robot
Greetings, 
Kuba

cambel, 2023-07-19:
Hi @PartyKusZ,

I have added support for the real Robotiq gripper. You need to checkout the latest commit of this repository and move to the branch `noetic-real-test` in the `robotiq` [package](https://github.com/cambel/robotiq/tree/noetic-real-test) 

Then you need to set the argument `gripper` of the `Arm` as `'85'`.

Try and let me know if you have problems with it. I have been using it and it works fine.

PartyKusZ, 2023-07-19:
I am not sure how to make this change:
"Then you need to set the argument gripper of the Arm as '85'."
Can you tell me how to make this change?
Thank you

cambel, 2023-07-20:
See this example
https://github.com/cambel/ur3/blob/b3e1bdcb783f0cf859cb3ea0f2c004d1c9eabb3a/ur_control/scripts/real_controller_examples.py#L119-L122

PartyKusZ, 2023-07-20:
after the changes I received errors : 
`rosrun ur_control joint_position_keyboard.py --namespace /ur3
Import ur_ikfast not available, IKFAST would not be supported without it
ft_sensor: False, ee_link: None, 
 robot_urdf: ur3e
connecting to: /ur3/scaled_pos_joint_traj_controller/command
[INFO] [1689848848.862789]: JointTrajectoryController initialized. ns: /ur3/
Traceback (most recent call last):
  File "/home/lab010/catkin_workspace/src/ur3/ur_control/scripts/joint_position_keyboard.py", line 218, in <module>
    main()
  File "/home/lab010/catkin_workspace/src/ur3/ur_control/scripts/joint_position_keyboard.py", line 211, in main
    ee_link=tcp_link)
  File "/home/lab010/catkin_workspace/src/ur3/ur_control/src/ur_control/arm.py", line 106, in __init__
    self._init_controllers(gripper, joint_names_prefix)
  File "/home/lab010/catkin_workspace/src/ur3/ur_control/src/ur_control/arm.py", line 131, in _init_controllers
    self.gripper = GripperController(namespace=self.ns, prefix=self.joint_names_prefix, timeout=2.0)
  File "/home/lab010/catkin_workspace/src/ur3/ur_control/src/ur_control/controllers.py", line 28, in __init__
    self.gripper_type = str(rospy.get_param(self.ns + node_name + "/gripper_type"))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/client.py", line 467, in get_param
    return _param_server[param_name] #MasterProxy does all the magic for us
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/msproxy.py", line 123, in __getitem__
    raise KeyError(key)
KeyError: '/ur3/gripper_controller/gripper_type'
`

PartyKusZ, 2023-07-20:
When i ran it:
`$ rosrun ur_control joint_position_keyboard.py`
i recived errors: 

`Import ur_ikfast not available, IKFAST would not be supported without it
ft_sensor: False, ee_link: None, 
 robot_urdf: ur3e
connecting to: /ur3/scaled_pos_joint_traj_controller/command
[INFO] [1689848848.862789]: JointTrajectoryController initialized. ns: /ur3/
`

`File "/home/lab010/catkin_workspace/src/ur3/ur_control/scripts/joint_position_keyboard.py", line 218, in <module>
    main()
`

` File "/home/lab010/catkin_workspace/src/ur3/ur_control/scripts/joint_position_keyboard.py", line 211, in main
    ee_link=tcp_link)`

`File "/home/lab010/catkin_workspace/src/ur3/ur_control/src/ur_control/arm.py", line 106, in __init__
    self._init_controllers(gripper, joint_names_prefix)
`

` File "/home/lab010/catkin_workspace/src/ur3/ur_control/src/ur_control/arm.py", line 131, in _init_controllers
    self.gripper = GripperController(namespace=self.ns, prefix=self.joint_names_prefix, timeout=2.0)`

`File "/home/lab010/catkin_workspace/src/ur3/ur_control/src/ur_control/controllers.py", line 28, in __init__
    self.gripper_type = str(rospy.get_param(self.ns + node_name + "/gripper_type"))
`

` File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/client.py", line 467, in get_param
    return _param_server[param_name] #MasterProxy does all the magic for us
`
`File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/msproxy.py", line 123, in __getitem__
 raise KeyError(key)
`

`KeyError: '/ur3/gripper_controller/gripper_type'
`
I changed the namespace directly in the code, from the namespace ur3 is running the ros drivers for ur3.
```python
    ns = '/ur3'
    joints_prefix = None
    robot_urdf = "ur3e"
    rospackage = None
    tcp_link = None
    use_gripper = args.gripper
    global arm
    arm = Arm(ft_sensor=False,
              gripper='85', namespace=ns,
              joint_names_prefix=joints_prefix,
              robot_urdf=robot_urdf, robot_urdf_package=rospackage,
              ee_link=tcp_link)
```

jabandersnatch, 2023-07-25:
Hi, first of all I'm really thankfull for all the development that you made for the UR3 robot, without you I could possibly be able to make my undergraduate project. I want to know If it's possible to run the hande robtiq gripper, with the UR3 and what are the steps to do that. 

I tried to run the ur_driver package to my robot to no avail.

cambel, 2023-07-26:
@PartyKusZ  
I just noticed that you are working on ROS Melodic.
I have not tested working with the real gripper on Melodic, it should work but you would need to adapt the code, I don't have the time right now to add compatibility to Melodic.

cambel, 2023-07-26:
Hi @jabandersnatch,

You could try to follow the steps that I mentioned in this issue to make it work with the Hand-e. It should work, but I haven't tested it and you would need to do some modifications. If you face any issue you can post it here, I can try to guide you.

On the other hand, I did try this other method a few years back.
Install this package and see the example in the script
https://github.com/cambel/ros_robotiq_urcap_control

PartyKusZ, 2023-07-26:
@cambel 
I informed you about the fact that I work in Melodic in the first message. 
Thank you for your help. 
Greetings

PartyKusZ, 2023-09-11:
Hi @cambel !
I'm back after the vacation :D Referring to our earlier conversation:
I installed ubuntu 20.04 lts and ros noetic. I tried to switch to the noetic-real-test branch, but from what I saw it is no longer there, I suspect that everything needed to move the actual gripper is on the noetic-devel branch. I changed the configurations in the ur3/ur_control/scripts/joint_position_keyboard.py:
```py
arm = Arm(ft_sensor=False,
              gripper='85', namespace=ns,
              joint_names_prefix=joints_prefix,
              robot_urdf=robot_urdf, robot_urdf_package=rospackage,
              ee_link=tcp_link)
```
 file but I still can't close the gripper.
First I run ```roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.10.200```
and then ```roslaunch ur_gripper_85_moveit_config start_moveit.launch config:=true```

Errors from ```roslaunch ur_gripper_85_moveit_config start_moveit.launch config:=true```
:
```[ERROR] [1694420332.871718631]: Unable to identify any set of controllers that can actuate the specified joints: [ robotiq_85_left_knuckle_joint ]
[ERROR] [1694420332.871739267]: Known controllers and their joints:
controller 'scaled_pos_joint_traj_controller' controls joints:
  elbow_joint
  shoulder_lift_joint
  shoulder_pan_joint
  wrist_1_joint
  wrist_2_joint
  wrist_3_joint

[ERROR] [1694420332.871757695]: Apparently trajectory initialization failed
```

Wranings from ```roslaunch ur_gripper_85_moveit_config start_moveit.launch config:=true```: 
```[ WARN] [1694420348.257971341]: The complete state of the robot is not yet known.  Missing robotiq_85_left_knuckle_joint```

Greetings

cambel, 2023-09-12:
Hi @PartyKusZ ,
you are right, I moved it to the noetic-devel branch.

Sorry, I forgot to mention that you need to add the setup of the gripper to any of your launch files
```
  <include file="$(find robotiq_control)/launch/urcap_cmodel_action_controller.launch">
    <arg name="address" value="192.168.10.200" />
    <arg name="config" value="cmodel_action_controller_85" />
  </include>
```

PartyKusZ, 2023-09-12:
@cambel 
I added the following code:
```
<include file="$(find robotiq_control)/launch/urcap_cmodel_action_controller.launch">
    <arg name="address" value="192.168.10.200" />
    <arg name="config" value="cmodel_action_controller_85" />
</include>
```
to ros_ur3/ur_gripper_85_moveit_config/launch/start_moveit.launch 

I get the same errors and warnings  as before from  ```roslaunch ur_gripper_85_moveit_config start_moveit.launch config:=true```and new one:```Failed to activate gripper in ns [/gripper/]```

Tell me please how you run ROS packages to move the real ur3 and gripper. Perhaps I was making some obvious mistake. Thank you for all your help. Greetings

PartyKusZ, 2023-09-18:
Hi @cambel 
I discovered a new problem. When I run the gripper directly in ur3 computer and then try to run ```roslaunch ur_gripper_85_moveit_config start_moveit.launch config:=true``` I get this error: ```Joint 'finger_joint' not found in model 'ur_robot_gazebo'```.
Greetings.

PartyKusZ, 2023-09-21:
@cambel 
I tested it. I got following error: 
```RLException: arg 'kinematics_config' is not defined. 

Arg xml is <arg name="kinematics_params" value="$(arg kinematics_config)"/>
The traceback for the exception was written to the log file
```

cambel, 2023-11-10:
@PartyKusZ 

I finally had time to test it on the real robot. 
I make several modifications so if you can try again, check the latest commit on the `real-robot` branch. 
Also you need to move the `robotiq` package to the `noetic-real-robot` branch so that they can work together.

PartyKusZ, 2023-11-17:
@cambel 
Thank you for your reply. I am currently busy, I will most likely have time in February. I will let you know when I start testing. All the best!

PartyKusZ, 2024-02-19:
Hi @cambel !
I successfully moved the gripper on a real robot (using ```roslaunch robotiq_control cmodel_simple_controller.launch ip:=<>```). I am very grateful to you for your help. I would still like to ask for help on one more issue. I also need to move the gripper from rviz level. From my findings, it seems that calling ```roslaunch ur3_description bringup_with_gripper_85.launch robot_ip:<>``` does not start the gripper controller. 

```[ INFO] [1708340733.412458857]: Loaded ur_robot_driver hardware_interface
[ INFO] [1708340733.476183134]: waitForService: Service [/controller_manager/switch_controller] is now available.
[ INFO] [1708340733.476204793]: Service available.
[ INFO] [1708340733.476215587]: Waiting for controller list service to come up on /controller_manager/list_controllers
[ INFO] [1708340733.476701977]: Service available.
[ INFO] [1708340733.620459650]: Robot mode is now RUNNING
[ INFO] [1708340733.623416388]: Robot's safety mode is now NORMAL
[INFO] [1708340733.668121]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1708340733.674974]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1708340733.676400]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1708340733.682434]: Loading controller: pos_joint_traj_controller
[INFO] [1708340733.684621]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1708340733.694203]: Loading controller: joint_state_controller
[INFO] [1708340733.762392]: Loading controller: joint_group_vel_controller
[INFO] [1708340733.770368]: Loading controller: scaled_pos_joint_traj_controller
[INFO] [1708340733.778281]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[INFO] [1708340733.802272]: Loading controller: speed_scaling_state_controller
[INFO] [1708340733.810184]: Loading controller: force_torque_sensor_controller
[INFO] [1708340733.818436]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1708340733.826499]: Started controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
``` 

when calling rviz ```roslaunch ur_gripper_85_moveit_config start_moveit.launch config:=true>``` gets this error 

[ERROR] [1708340978.550176637]: Action client not connected: gripper_controller/gripper_cmd

I also noticed that by running the simulation ```roslaunch ur3_gazebo ur_gripper_85_cubes.launch ur_robot:=ur3 grasp_plugin:=1```

the controller from the grasp launches. All the while trying to trace the difference between launching controllers in a real robot call and a simulation call. 

Thank you very much for your help so far!