cambel/ur3 #28: [ WARN] ： The complete state of the robot is not yet known.  Missing hande_left_finger_joint

NIU55666, 2023-08-07:
When I run `roslaunch ur_hande_moveit_config start_moveit.launch`, a warning appears: "The complete state of the robot is not yet known. Missing hande_left_finger_joint." It keeps popping up. How can I resolve this?

cambel, 2023-08-08:
@NIU55666 
What version of ROS are you working on?
Are you using docker or installing from source?

NIU55666, 2023-08-08:
> 

> @NIU55666 What version of ROS are you working on? Are you using docker or installing from source?

My ROS version is Noetic. I'm not using Docker; I'm installing it from source.

cambel, 2023-08-15:
@NIU55666 

I can't reproduce what you described. Can you provide more information on what you are trying to do and the trace from the terminal?

Also, make sure that you are on the latest commit of the robotiq package.
I uploaded some changes to this `ur3` repo, you may consider reinstalling/building it

closed 2023-08-18 by cambel