cambel/ur3 #29: Connection refused with UR5e and ROS noetic

Yeesha-R, 2024-03-27:
### Summary
Connection to remote PC could not be established

I have been experiencing issues connecting to my UR5e lately (it started few days ago). I read all issues under "connection refused". I have also read issues about not using the correct IP address, in issues [#244,](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/244) [#590](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/590), [#510](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/510). However, none of the solutions worked for my problem. Here is my setup:

### Version
ROS Driver version: Ubuntu 20.04, ROS noetic
Affected Robot Software Version(s): e-Series
Affected Robot Hardware Version(s): UR5e
URCaps Software version(s): externalcontrol-1.0.5.urcap

### Issue 
When I run `roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.1.2`, I get the following output:

```
... logging to /home/ubuntu/.ros/log/c03e5a20-ec68-11ee-bf0d-3b86055ec174/roslaunch-yeesha-55242.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.1:43959/

### SUMMARY
PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /forward_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_cartesian_traj_controller/type: pass_through_cont...
 * /forward_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_joint_traj_controller/type: pass_through_cont...
 * /hardware_control_loop/loop_hz: 500
 * /joint_based_cartesian_traj_controller/base: base
 * /joint_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /joint_based_cartesian_traj_controller/tip: tool0
 * /joint_based_cartesian_traj_controller/type: position_controll...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 500
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /pose_based_cartesian_traj_controller/base: base
 * /pose_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /pose_based_cartesian_traj_controller/tip: tool0_controller
 * /pose_based_cartesian_traj_controller/type: pose_controllers/...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /scaled_pos_joint_traj_controller/action_monitor_rate: 20
[... 259 lines cut ...]

I was able to successfully ping the robot at 192.168.1.2 
[image]

When I ran `telnet 192.168.1.1 50002`, I got 
```
Trying 192.168.1.1...
telnet: Unable to connect to remote host: Connection refused
```
When I ran `nc 192.168.1.2`, I got 
`nc: missing port number`

when I ran `sudo ufw disable`, I got `Firewall stopped and disabled on system startup` 
and when I ran  `ss -alpn | grep 5000 | column -t`, I noticed that the 50002 port is not among the available ports
```
tcp  LISTEN  0  1  0.0.0.0:50001  0.0.0.0:*  users:(("ur_robot_driver",pid=56686,fd=11))
tcp  LISTEN  0  1  0.0.0.0:50003  0.0.0.0:*  users:(("ur_robot_driver",pid=56686,fd=19))
tcp  LISTEN  0  1  0.0.0.0:50004  0.0.0.0:*  users:(("ur_robot_driver",pid=56686,fd=22))
```

A screenshot is attached below
[image]

I have been using this robot for over one year but this issue started recently. 
Kindly let me know if you need anymore information. Thank you.

cambel, 2024-03-28:
Hi @Yeesha-R 
The issue is unrelated to this repository. I recommend you open one directly on the [Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) repository.
That being said, the problem seems to be the network configuration between the robot and the PC. So you may want to look into ways to open the required ports for the robot to properly communicate with the PC

Yeesha-R, 2024-04-01:
Hi @cambel ,

Thank you for your response. It was a mistake and I tried to delete it but could not. Thanks for the suggestion, I fhave found a solution to the issue.

closed 2024-04-01 by Yeesha-R

Whohahaha, 2024-04-09:
> Hi @cambel ,
> 
> Thank you for your response. It was a mistake and I tried to delete it but could not. Thanks for the suggestion, I fhave found a solution to the issue.

may your experiences and solution shared? I am facing the same problem as you!

Yeesha-R, 2024-04-09:
@Whohahaha , please see issue https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/690 on the UR robot's github page

Whohahaha, 2024-04-10:
thanks so much!